limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera #6: `Failed to get camera frame buffer`

mrusme, 2024-01-31:
I keep running into the following line:

https://github.com/limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera/blob/78ad1b1d80888eb9d571d5725a91b527469fac79/take_photos/take_photos.ino#L20

What could be the reasons why the frame buffer could not be grabbed? I'm trying to rule out a hardware defect. Initialization seems to work flawlessly, taking a photo however seems not possible. Using the following settings:

```sh
arduino-cli compile --fqbn esp32:esp32:XIAO_ESP32S3 --clean -e ./
Sketch uses 947797 bytes (28%) of program storage space. Maximum is 3342336 bytes.
Global variables use 50076 bytes (15%) of dynamic memory, leaving 277604 bytes for local variables. Maximum is 327680 bytes.

Used library     Version Path
I2S              1.0     /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/I2S
FS               2.0.0   /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/FS
SD               2.0.0   /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SD
SPI              2.0.0   /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/SPI
Base64           1.0.0   /home/mrus/.local/share/arduino/libraries/Base64
HTTPClient       2.0.0   /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/HTTPClient
WiFi             2.0.0   /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/WiFi
WiFiClientSecure 2.0.0   /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14/libraries/WiFiClientSecure

Used platform Version Path
esp32:esp32   2.0.14  /home/mrus/.arduino15/packages/esp32/hardware/esp32/2.0.14
```

```sh
arduino-cli upload -p /dev/ttyACM1 --fqbn esp32:esp32:XIAO_ESP32S3 --input-dir ./build/esp32.esp32.XIAO_ESP32S3/
esptool.py v4.5.1
Serial port /dev/ttyACM1
Connecting...
Chip is ESP32-S3 (revision v0.2)
Features: WiFi, BLE
Crystal is 40MHz
MAC: XX:XX:XX:XX:XX:XX
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000f7fff...
Compressed 15104 bytes to 10399...
Writing at 0x00000000... (100 %)
Wrote 15104 bytes (10399 compressed) at 0x00000000 in 0.2 seconds (effective 518.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.0 seconds (effective 576.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 767.3 kbit/s)...
Hash of data verified.
Compressed 948224 bytes to 616325...
Writing at 0x00010000... (2 %)
Writing at 0x0001b302... (5 %)
Writing at 0x00025bb0... (7 %)
Writing at 0x0002f29a... (10 %)
Writing at 0x0003e07e... (13 %)
Writing at 0x00043b31... (15 %)
Writing at 0x000494ef... (18 %)
Writing at 0x0004f0bf... (21 %)
Writing at 0x00054884... (23 %)
Writing at 0x0005a23f... (26 %)
Writing at 0x0005f40f... (28 %)
Writing at 0x000644c4... (31 %)
Writing at 0x000693ec... (34 %)
Writing at 0x0006e34d... (36 %)
Writing at 0x00073305... (39 %)
Writing at 0x00078123... (42 %)
Writing at 0x0007d801... (44 %)
Writing at 0x000840ec... (47 %)
Writing at 0x000892ed... (50 %)
Writing at 0x0008e2bb... (52 %)
Writing at 0x000930c3... (55 %)
Writing at 0x00098397... (57 %)
Writing at 0x0009df68... (60 %)
Writing at 0x000a31b4... (63 %)
Writing at 0x000a83cc... (65 %)
Writing at 0x000ad244... (68 %)
Writing at 0x000b22f8... (71 %)
Writing at 0x000b74de... (73 %)
Writing at 0x000bc7b3... (76 %)
Writing at 0x000c1caf... (78 %)
Writing at 0x000c7a35... (81 %)
Writing at 0x000ccf23... (84 %)
Writing at 0x000d244d... (86 %)
Writing at 0x000da806... (89 %)
Writing at 0x000e36d3... (92 %)
Writing at 0x000e8ae9... (94 %)
Writing at 0x000ee5da... (97 %)
Writing at 0x000f3ea9... (100 %)
Wrote 948224 bytes (616325 compressed) at 0x00010000 in 7.0 seconds (effective 1082.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
New upload port: /dev/ttyACM1 (serial)
```

limengdu, 2024-02-02:
Have you enabled the PSRAM function on the XIAO ESP32S3? If it is not turned on, the camera will not work properly.

mrusme, 2024-02-02:
@limengdu I'm using the following commands:

```sh
arduino-cli compile --fqbn esp32:esp32:XIAO_ESP32S3 --board-options PSRAM=opi --clean -e ./
```

```sh
arduino-cli upload -p /dev/ttyACM1 --fqbn esp32:esp32:XIAO_ESP32S3 --board-options PSRAM=opi --input-dir ./build/esp32.esp32.XIAO_ESP32S3/
```

So I would assume that `PSRAM` is set to `opi`. Do you happen to know a way to check?

limengdu, 2024-02-04:
There doesn't seem to be any problem. You are using this code with both the SD card and the camera, did you cut the pads as per the Wiki instructions? Or is the camera working properly when you run the camera's program alone?

mrusme, 2024-02-05:
J3 doesn't appear to be connected:

[image]

limengdu, 2024-02-06:
Everything looks good. Does running the camera programme alone work?

mrusme, 2024-02-09:
To give a quick status feedback: [This code](https://github.com/limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera/tree/main/take_photos) works and it can successfully retrieve the frame buffer to take a photo. I have pretty much copied it into my code and have been testing ever since, with little to no success. I'm still debugging the issues, but I get the feeling that ...

- ... either the microphone (`I2S.read()`) is interfering
- ... or the WiFi connection that I'm using is interfering
- ... or the `esp_sleep_*` feature is interfering

I'm going to continue to debug the issue and will report back.

mrusme, 2024-02-09:
Update:

>     * ... either the microphone (`I2S.read()`) is interfering

This appears to be the case. The moment I enable this piece of code, the camera stops working:

```c
void initMic() {
  I2S.setAllPins(-1, 42, 41, -1, -1);
  if (!I2S.begin(PDM_MONO_MODE, 16000, 16)) {
    Serial.println("Failed to initialize I2S!");
    while (1);
  }
}
```

limengdu, 2024-02-18:
It looks like there is something wrong with the microphone, have you tried the programme with the microphone alone?

closed 2024-05-17 by limengdu