limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera #8: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

hpsaturn, 2024-04-21:
## Summary

I was testing this Camera with its official samples, but the Camera presents aleatory behavior, sometimes it works fine for some minutes, but stop of working. Initial I used this code:

https://github.com/hpsaturn/esp32s3-cam/blob/master/examples/xiao-espnow-sender/xiao-espnow-sender.cpp

but I also returned to the official code of this repo but nothing. With other Cameras like the Freenove S3 Camera, everything works fine. With this module, only glitches and lost time. 

## Questions

- Is necessary an Especial version of Espressif Framework? or the Arduino IDE framework?  
- Is possible that the Sensor was damaged? I tested it and worked for long time (10 minutes), that I mentioned, but it getting very hot.

## Environment

```python

Platform espressif32 @ 6.6.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20014.231204 (required: platformio/framework-arduinoespressif32 @ ~3.20014.0)
├── tool-esptoolpy @ 1.40501.0 (required: platformio/tool-esptoolpy @ ~1.40501.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-openocd-esp32 @ 2.1100.20220706 (required: platformio/tool-openocd-esp32 @ ~2.1100.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5)
└── toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5)
```

## Steps to replicate

I'm using PlatformIO, for running it, put the next `platformio.ini` file into the directory `CameraWebServer` for example, and running `pio run --target upload`

```python
; ESPNowCam Freenove ESP32S3CAM
; https://github.com/hpsaturn/esp32s3-cam
; @Hpsaturn 2024

[platformio]
src_dir = ./

[wifi]
ssid = ${sysenv.PIO_WIFI_SSID}
password = ${sysenv.PIO_WIFI_PASSWORD}

[env]
platform = espressif32
framework = arduino
monitor_speed = 115200
monitor_filters = 
  esp32_exception_decoder
  time
build_flags =
  '-DWIFI_SSID="${wifi.ssid}"'
  '-DWIFI_PASS="${wifi.password}"'
  -D CORE_DEBUG_LEVEL=4
  -D BOARD_HAS_PSRAM=1

[esp32common]
extends = env
board = esp32-s3-devkitc-1
board_build.flash_size = 8MB

[env:xiao]
extends = esp32common
board_build.arduino.memory_type = dio_opi ;
```

beniroquai, 2024-04-26:
This looks like a hardware issue to me. 
My pio config for the xiao is the following https://github.com/Matchboxscope/matchboxscope-simplecamera/blob/matchboxscope/platformio.ini#L33

hpsaturn, 2024-04-26:
Trying to validate if the camera module is damaged, I installed this firmware that you recommended:

https://matchboxscope.github.io/firmware/FLASH.html

but I have the same behavior:

```cpp
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x420cd432
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
[   253][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[   258][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: serialFrame NOT_FOUND
Serial transfer mode disabled
[   259][D][main.ino:249] setNeopixel(): Setting Neopixel to 0
[   261][D][main.ino:249] setNeopixel(): Setting Neopixel to 255
[   326][D][main.ino:249] setNeopixel(): Setting Neopixel to 0
Starting internal SPIFFS filesystem
Listing SPIFFS directory: /
Xiao Sense Camera initialization
[  1039][V][Preferences.cpp:388] getInt(): nvs_get_i32 fail: serialFrame NOT_FOUND
[  1039][I][main.ino:525] StartCamera(): Serial Frame Disabled
E (800) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)

CRITICAL FAILURE: Camera sensor failed to initialise.

A full (hard, power off/on) reboot will probably be needed to recover from this.
Meanwhile; this unit will reboot in 1 minute since these errors sometime clear automatically
E (10911) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
[ 11164][D][main.ino:591] StartCamera(): Second attempt worked..
[ 11165][E][main.ino:692] StartCamera(): Camera frame error
[ 11170][E][main.ino:692] StartCamera(): Camera frame error
[ 11175][E][main.ino:777] setup(): Camera failed to initialize 0
```

hpsaturn, 2024-04-27:
well, I fixed the issue changing the sensor in the camera by other, from the Freenove S3 Camera, and the XIAO is working. Mmm bad quality materials here? I don't know, but is working:

https://github.com/limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera/assets/423856/1c974ecf-5cfa-498c-bcf5-770c4c00179f

closed 2024-04-29 by limengdu