limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera #9: ledcSetup' was not declared in this scope;

samuk, 2024-06-10:
/home/sam/Desktop/SeeedStudio-XIAO-ESP32S3-Sense-camera-main/CameraWebServer/app_httpd.cpp: In function 'void setupLedFlash(int)':
/home/sam/Desktop/SeeedStudio-XIAO-ESP32S3-Sense-camera-main/CameraWebServer/app_httpd.cpp:1390:5: error: 'ledcSetup' was not declared in this scope; did you mean 'ledc_stop'?
 1390 |     ledcSetup(LED_LEDC_CHANNEL, 5000, 8);
      |     ^~~~~~~~~
      |     ledc_stop
/home/sam/Desktop/SeeedStudio-XIAO-ESP32S3-Sense-camera-main/CameraWebServer/app_httpd.cpp:1391:5: error: 'ledcAttachPin' was not declared in this scope; did you mean 'ledcAttach'?
 1391 |     ledcAttachPin(pin, LED_LEDC_CHANNEL);
      |     ^~~~~~~~~~~~~
      |     ledcAttach

exit status 1

Compilation error: 'ledcSetup' was not declared in this scope; did you mean 'ledc_stop'?

beniroquai, 2024-06-10:
Did you switch to Arduino-IDF 3.0?

> Message ID: <limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera/issues/9@
> github.com>
>

samuk, 2024-06-10:
Yes. I've since downgraded to 2.xx but it fails to work with an unrelated problem

closed 2024-06-10 by samuk

limengdu, 2024-06-11:
The esp-arduino version 3.0.x updated many interfaces that are no longer supported.

knmcguire, 2024-07-26:
just for some more context, I see this happen too with Arduino IDE 2.3.2

knmcguire, 2024-07-26:
but it seems that the CameraWebServer_for_esp-arduino_3.0.x.ino seems to do the trick! it works for me

AliBas-sudo, 2025-01-24:
Ok but what do you mean by 'CameraWebServer_for_esp-arduino_3.0.x.ino '  ? is it a file somwhere ? 
Currently I had the same issue after porting my code into Arduino IDE 2.3.2

limengdu, 2025-01-24:
https://github.com/limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera/blob/main/CameraWebServer_for_esp-arduino_3.0.x/CameraWebServer_for_esp-arduino_3.0.x.ino
@AliBas-sudo