limengdu/SeeedStudio-XIAO-ESP32S3-Sense-camera #12: Missing header files?

MkMunich, 2024-07-17:
Hi,
where do I find the additional header files, which are included in the source code? e.g. esp_camera.h and img_converters.h
Best regards
MK_Munich

limengdu, 2024-07-22:
Hi, when you select the supported ESP32 development board, the board comes with the file esp_camera.h, you don't need to add it additionally. As for the img_converters.h you mentioned, which code is this inside? Is it possible to provide a link?

MkMunich, 2024-07-25:
Hi, my mistake. After choosing board XIAO_ESP32S3 I was able to compile. The douwnload started, but stops after 40% - refer to attached file.
Do you know, why?

[XIAO_ESP32S3 Download Error.txt](https://github.com/user-attachments/files/16379828/XIAO_ESP32S3.Download.Error.txt)

limengdu, 2024-07-26:
MACOS？

MkMunich, 2024-07-29:
Windows 11
Arduino IDE 2.3.2
Board: Arduino ESP32 Boards 2.0.13
Libraries: non

limengdu, 2024-07-30:
```
A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'Das Ger t erkennt den Befehl nicht.', None, 22)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
```
This is the key part of your error message, it looks like it's a problem of not installing the esp-arduino on-board package completely, I would suggest you to REMOVE the esp on-board package first and then reinstall it completely again, or update to 2.0.14.

closed 2024-07-30 by limengdu